at67/ROMvX0 #31: Wrong iterations in nested for loops

axelb68, 2022-12-11:
The following code leads to a wrong number of iterations for the main loop. This is also observed when using the **call** command instead of **gosub** ( changing label p to proc p ).

`n = 5
m = 10
for i = 1 to n
	gosub p
next i
print i						'shows m+1  !   
END
p:
	for j = 1 to m
	next j	
return`

axelb68, 2022-12-11:
alternative formatting:
```
n = 5
m = 10
for i = 1 to n
	gosub p
next i
print i						'shows m+1  !	   
END
p:
	for j = 1 to m
	next j	
return
```

at67, 2022-12-12:
Cheers, checking it out now.

at67, 2022-12-12:
Ooof, I've found the bug:

- Currently there is a stack of zero page variables allocated for non-simple nested 'for loops', (i.e. for loops that use vars), the logic that controls this stack never considered that for loops could be called within GOSUBS or PROCS, so it resets the stack index when entering the GOSUB thereby reusing stack space it shouldn't be.
- I'll have to think about this one a bit to come up with a better solution, for now don't do it :)

at67, 2023-06-17:
A stack frame has been added for nested for loops vars, (in the next release), this should solve all of the above.