at67/ROMvX0 #35: Local variable changed by called procedure

axelb68, 2023-03-29:
```
_runtimePath_  "runtime"
_runtimeStart_ 0xFFFF
_arraysStart_  0xFFFF
_codeRomType_  ROMvX0

CLS
call p1
END

proc p1
local y
    x = 0
    call p2, x
    print x         ' prints 0
    y = 0
    call p2, y
    print y         ' prints 1 !
endproc

proc p2, a
    a = 1
endproc
```

A local variable within a procedure can be changed by a second procedure when used as its parameter.

at67, 2023-04-01:
This is by design and something I have never fully documented or standardised, let me explain how the system works first:

1) There are eight 16bit zero page slots allocated for both proc params and for proc local vars.
2) You can have 8 proc params or 8 local vars or anything in-between.
3) You can pass proc params by reference by knowing which slot numbers have been used by which params and which local vars.
4) In your example "y" gets passed by reference as it is using the same slot number as "a".
5) This is a confusing and adhoc system with no obvious syntax that decribes what is going on.
6) If you look at Invader in the gbas/games folder and check out "proc getNextInvaderShot", you'll see how I take advantage of it.
7) I need to come up with an obvious system that makes it clear what is going on.
8) A future version of ROMvX0 will handle this better, but I'd like to make it backwards compatible and clearer in the current language.
9) I was hoping no one would find this until I had solved it myself :)

Any thoughts?

axelb68, 2023-04-03:
The compiler could generate a warning in such cases, e.g. "y passed by reference". The warning may be avoided by adding a reference operator (ref y or &y).
Cases in which the local variable slot and parameter slot do not match or if one tries to pass a global variable by reference would cause an error message.

at67, 2023-06-17:
The next release, (in the next few days), uses a stack frame for locals and params.
References, (for now), no longer exist, but you can now return a result from a PROC.